first95/FRC2018 #12: Add basic CANTalon features to DrivePod class

jwalthour, 2017-10-11:
Specifically:
- Set current limit, and confirm that it works
- Set voltage limit, and confirm that it works
- Set voltage compensation

lindsayvallen, 2017-11-29:
@RoboticsNA 
Talon software manual is at http://www.ctr-electronics.com/downloads/pdf/Talon%20SRX%20Software%20Reference%20Manual.pdf

Current limit is discussed on page 65, voltage compensation on page 64, and I didn't find voltage limit yet.  It's possible voltage compensation can not only scale up but also down and hence might do voltage limiting as well.

RoboticsNA, 2017-12-07:
The built-in CANTalon voltage compensation does not work at the moment and for the time being it is set as an adjustedTalon.

RoboticsNA, 2017-12-07:
SOME NOTES!:

CurrentLimiting seemed to work:
(How To Set)
       leader.setCurrentLimit(numeric current threshold "in amps"); 
       leader.EnableCurrentLimit(boolean);

VoltageLimiting was not found:
(No .setVoltageLimit is in the talon features)

Voltage Compensating didn't work:
       leader.changeControlMode(TalonControlMode.Voltage); //Set talon to volt comp mode
       leader.setVoltageCompensationRampRate(24.0) //This means 0V to 12V in 50ms
       leader.set(6.0) //This sets the target output voltage

(Volt comp mode is completed by sampling the battery voltage and scaling the output duty cycle to match the desired target output voltage.)
(When testing this the robot went REALLY slow, as if it was limiting the voltage. I attempted to change the .set and .setVoltageCompensationRampRate values really big and really samll, as well as a bunch of numbers in between. Nothing seemed to work, the robot looked as it it went at the same speed. When the volt comp mode is enabled the TalonOutputVoltage gets to 1 and -1. With volt comp mode off the TalonOutputVoltage gets to high values such as 4 and -4)

(I tried testing with the current limiting on and off, I also tested with setting both the leader talon and the follower talons, it still didn't work.)

lindsayvallen, 2018-01-18:
Note that this work has changed with the 2018 code.  For specifics, see [CTRE migration guide](https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/Migration%20Guide.md) and [CTRE 2018 documentation](https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md).

Some specifics:

- voltage compensation is no longer a separate control mode, but rather an option that can be set in addition to another control mode.
- the current limit can be set as one value for continuous and another value for peak

jwalthour, 2018-01-29:
I think we can call this done, unless we find ourselves wanting more of these later.  The AdjustedTalon class is being used to compensate for voltage at the moment.

closed 2018-01-29 by jwalthour