first95/FRC2018 #13: Add closed-loop distance control mode to DrivePod class

jwalthour, 2017-10-11:
This is the capability to command the drive wheels to go X inches along the floor, using the PID loop on the Talon.

lindsayvallen, 2018-01-10:
I recommend trying the closed-loop distance control work first with the Talon fixture that has a rotational arm and only afterwards with Orville because Orville has slipping issues.

jwalthour, 2018-01-29:
Relies on #31 for tuning

lindsayvallen, 2018-02-03:
Progress made on this item in branch feature/drivepod_position_close_loop.  None of it tested yet.

lindsayvallen, 2018-02-09:
As of commit 902056e, pushing drive fwd or drive back buttons actually did something, but it made the robot spin in circles very fast, even after disabling and re-enabling.  Fixed assumed sign issue in commit 6bbb14e and made encoder ticks per inch more reasonable in 2c77efe, but still goes in circles, albeit not quite as fast of ones.  Need to debug on the robot with the encoder counts opened as plots to see what's happening there.  @jwalthour could you take a quick look at this branch to see if you observe an obvious issue?

jwalthour, 2018-02-09:
I don't see any obvious issues; everything looks like it oughta work.

One thing I did notice with the elevator: those sound like the symptoms I observed when the encoder was oriented in the opposite direction from what the Talon was expecting.  I only figured that out because the graph of the measured position was moving in an exponential curve (like a trace of a y = x^k kind of shape).  I wound up trying a call to `setSensorPhase(true)`, and it magically started working.  I believe the output can be inverted with `setInverted(true)` if that's what's needed.

jwalthour, 2018-02-09:
Just got a chance to take a look at this on the practice robot.  It looks like it was just the phase!  I'll try to refine it a little bit tonight.

jwalthour, 2018-02-10:
As of 8ac94ee9354cf1fe7aa95adf2b95dc41dad2c46d, the drive pods are doing some sort of closed-loop position control.

However, neither side accepts a command to seek a negative position.  Soft limits are disabled, so I don't think it's that.  It's possible that an exception is being thrown, and silently caught someplace, but nothing shows up in the log.  This issue affects both sides equally.

[image]

lindsayvallen, 2018-02-13:
As of b8795f0, fixed this bug where target wouldn't go below 0.  Now doing PID tuning.

lindsayvallen, 2018-02-16:
Works to move 4' with only a 1" error, which I'm calling good enough for now.  We can do additional tuning when we do the competition robot.  Finished and merged into master in commit 99c425b.

closed 2018-02-16 by lindsayvallen