first95/FRC2018 #14: Add closed loop speed control to the DrivePod class

jwalthour, 2017-10-11:
This is the feature where we tell the DrivePod to maintain X inches per second along the carpet, using the PID controller in the Talons.

lindsayvallen, 2018-01-10:
This may be easier to test first with the Talon connected to the rotational arm and then when it's worked out there, with Orville.

jwalthour, 2018-01-29:
Relies on #31 for tuning

lindsayvallen, 2018-02-14:
@jwalthour What are we using this for?  I think we want to use closed-loop position control for auto moves, and if that doesn't work well, then motion profiling.  I don't see a main use case for closed-loop speed control of the drivebase.  If you don't either, then we should move this issue to the Useful Supporting Features milestone.

jwalthour, 2018-02-14:
It'd be for teleoperated driving, but now that I think about it, I'm not sure how much need there is.  The main idea is to compensate for random stuff that slows down the drivetrain.  Kovaka was the last robot that actually had a need for this - the 2017 nor 2018 designs both have strong mechanical forces keeping the drivebase moving straight forward.  With Kovaka, if you pushed the stick straight forward, sometimes the bot would curve to the right.  But now that we've seen the 2018 robot driving, it doesn't look like it even remotely has that problem.

We can probably move it into nice-to-haves, or mark it "wontfix" and close it.

closed 2018-02-14 by jwalthour