first95/FRC2018 #35: Add support for collector subsystem

jwalthour, 2018-01-29:
This includes:
* Intake on/off/reverse
* Maw open/shut
* Set the intake to one of its 4 available heights

jwalthour, 2018-01-30:
We've got everything except height as of 807d692e8c640fbdbd8a6b5b46c63ad651e72e54

lindsayvallen, 2018-01-30:
The height for collector is controlled by a pair of pistons (solenoids) where the four possible combinations of their Boolean values sets one of four local heights of the collector.  The more global height of the collector is determined by the elevator.

lindsayvallen, 2018-01-30:
Height for collector is a local height that is set to one of four positions by a pair of solenoids.  The global height for the collector is determined by the elevator, which is a separate subsystem.

lindsayvallen, 2018-02-01:
@RoboticsNA added the height option ("wrist" of the collector) and tested it in commit cbd2da9.  Still manual control.

closed 2018-02-03 by jwalthour