first95/FRC2018 #54: Add automatic cube acquisition

jwalthour, 2018-02-12:
This would be useful both as a standalone auto move, and as a mode to be used in teleoperated mode.

There will be 3 optical proximity sensors mounted on the rear plate of the maw - left, center and right.  They'll all be set to the same range, but the cube's orientation will affect which ones trip.  The geometry is set up so that the cube can't trigger two of them until it's deep enough into the maw to be grabbed.

 The acquisition algorithm shall be as follows:
* Wait until at least two of the sensors report a bright object within range.
* Close the maw
* Run the chains inward until some condition is satisfied.  We're not sure yet what the best condition is.  Some ideas to try:
    * The current draw of the chain drive motors spikes, indicating extra drag
    * A certain amount of time has elapsed (like 1/3 of a second)
    * All 3 sensors are tripped

lindsayvallen, 2018-02-15:
@jch13 suggested this have only slightly lower priority than auto moves.  As of commit 99a3029, the first two parts of this move work as an auto move -- open maw, wait until at least two sensors report object in range, close maw.

jwalthour, 2018-02-17:
WOrks as of c1dab2dbacbab6ae2884b6a253d5e27b3b503d93

closed 2018-02-17 by jwalthour