first95/FRC2018 #58: Kepler drives backwards

jwalthour, 2018-02-19:
You may have already seen Jame's notes on Slack indicating that Kepler (the new robot) seems to be running the drivebase and elevator backwards compared to Doppler (the practice robot).  James suspects that it's just the talons numbered incorrectly - the ones on the left bearing the addresses that the software considers the right.  I have a sneaking suspicion that Kepler could have the addresses in the correct locations, and that Doppler could be the one that's backwards.  In any case, I think these are the things that we can check:

- Look at the printed labels on the Talons on both Kepler and Doppler, and make sure the labels match the assignments in Constants.java.  If this is incorrect, I recommend leaving the talons physically in place, and changing their addresses with the RoboRIO web dashboard.  Don't forget to fix the printed labels as well to match.
- If those are both correct, then for each drive pod, check that the wire color mapping is the same.  The Talon outputs are white and green wires, but the drive motors have red and black wires.  I believe we want both robots to be white-to-red and green-to-black.  If you find one robot wired with one color scheme and another wired with the reverse, check with James to find the correct mapping.
- If those are both correct, it's time to use the RoboRIO web dashboard to make sure the Talons have the same address internally as each printed label.  To do this:
    1. Connect your computer to the robot with the USB cable
    2. Open Internet Explorer and connect to http://172.22.11.2/
    3. You should soon see a list of Talons, and their addresses.
    4. To locate a talon, check the "blink LED" box and then hit "save".  (It won't blink until you hit save).
    5. To move a Talon to a new address, change its address and click "save".  If you get two talons on the same address, it's not a problem - command it to change address, and one of them will get the new address.

jwalthour, 2018-02-19:
Cause discovered - Kepler's addresses are assigned backwards. 
[image]

RoboticsNA, 2018-02-19:
The talons were switched in the web console, issue fixed!

closed 2018-02-19 by RoboticsNA

jwalthour, 2018-02-19:
Don't forget to update the physical labels as well!  That will save future confusion.