first95/FRC2018 #59: Sweeper turns don't work at all

jwalthour, 2018-02-19:
As of d1ce6557400921dd630431d2e21533015737d649, the sweeper turns really don't sweep.  They cause the robot to drive straight, and then pivot at the end.  I believe this is because we're using distance closed-loop control to command a particular distance on each side, so the talons just drive both sides forward at a similar rate.

jwalthour, 2018-02-19:
An attempted fix exists in feature/speed_CL_control as of 6e0e90b006ac6b142ab86b0a8bdde977f6f0fe72.  Need to test; probably wont' work the first time

jwalthour, 2018-02-21:
I think a big part of the problem is that the individual pods aren't reaching their commanded speeds.

These graphs are from commit 24e112754e5833bbdff79c649460ef7921ff4a9f, after commanding "drive forward 2 feet in 4 seconds".  Note that the speed target is correct (6" each side) but the attained speed remains much lower (between 4 and 4.5 inches per second).  This is consistent with the behavior that we're seeing from the sweeper turns - far too wide of a turn.  Also I think there's something wrong with how the targets are getting computed, because it no longer stops on its own.  I probably mucked that up when I fixed the pivots to use speed-based control.

[image]

lindsayvallen, 2018-04-07:
This would be really good to get worked out for future years, but we're far enough away on this issue and pivot turns work well enough that I'm reclassifying this issue as "Maybe Later"