first95/FRC2018 #60: Pivot turns aren't consistent

jwalthour, 2018-02-19:
Commanding a pivot of 90 degrees will often result in a pivot of something visible more or less than 90 degrees.  This behavior is present as of d1ce6557400921dd630431d2e21533015737d649, and seems to happen similarly on concrete and on carpet.

lindsayvallen, 2018-02-20:
Tested at commit 37d8652 on concrete floor and carpet.  On concrete floor, turns seemed to be less than 90 degrees but somewhat consistent.  On carpet, turns seemed to be less than 90 degrees, generally, but much less consistent.  Uncertain to what extent there are friction/slip issues on carpet that will be there on field that we need to address vs. issues on our carpet due to bunching.  Possible that sweeping turns might give us more consistency so suggest focusing on issue #59 first, and if we get those working well, then drop this issue.  If those don't work, then re-visit this.

lindsayvallen, 2018-03-10:
Pivot turns are working somewhat more consistently now, with the speed CL control branch work (commit 3c5c9dc).  They're not getting to quite the right orientation, so adding Pigeon IMU to get heading information.  Once that's done in #75, then we need to incorporate Pigeon IMU heading information to close the control loop on heading with pivot turns.

jwalthour, 2018-03-24:
Finally done in d075e01d6e99a2c00052cc455d5485c10eeb926f!  This was solved by adding a second level of PID control, which takes robot heading as input and outputs desired pivot rate.

closed 2018-03-24 by jwalthour