first95/FRC2018 #61: PID Controlling on Kepler's Elevator doesn't work

RoboticsNA, 2018-02-19:
James commented that none of the PID tuning or control doesn't work on Kepler.

Possible encoder issue?

jwalthour, 2018-02-19:
What're the symptoms?

One thing worth checking early on: is the encoder connected to the same talon that is assigned as the leader?

RoboticsNA, 2018-02-19:
It works now, the talon from Doppler (Talon 23) was put onto Kepler. This made Kepler work but not sure if Doppler works. Will test and report here.

lindsayvallen, 2018-02-24:
Tested Doppler today after swapping out its winch talons (13 and 23) and updating their firmware, but still having issue where can't maintain position.  If moved to a particular position, get print statement saying maintaining position, but just falls back home.  There were some mechanical issues with slippage that were fixed, but problem persists.

lindsayvallen, 2018-02-24:
Did a bunch of debugging and couldn't resolve issue from code.  Un-bagged Kepler and took screenshots of its talons settings from the web config.  Appeared that the PID settings for 23 had non-zero values for Kepler and zero values for Doppler so changed Doppler's to match.  Also fixed some of the CAN wiring, and swapped Roborios with Orville (although don't think that was it).  Not sure which of these steps fixed the issue, perhaps putting the PID values in the web config for Doppler's talon 23, but something did.

closed 2018-02-24 by lindsayvallen

jwalthour, 2018-02-24:
One cause of this issue came about in  c4957a305f57d8eb7832448e8decc51fcd4cf5e6, where the setters for these constants was removed from the Elevator constructor.  We didn't notice it at the time, because the Talons remember these constants in nonvolatile memory.  But when we switched the talons to deal with CAN issues, the new Talons didn't have PID constants, bringing about the second prerequisite of this issue.  

In bfd9978f02d6011d11e7790047a7bfa50dce6f36 , the setters are added back in, which should keep this problem from happening again in the future.