first95/FRC2018 #63: When the elevator is going down with a cube it can get caught on elevator.

RoboticsNA, 2018-02-19:
When we have a cube and the wrist is in the upmost position and the elevator is going down it can sometimes get caught on the elevator *shaft* and this can break the motors. 

- Maybe make the wrist go down a notch when moving the elevator down in certain conditions?

RoboticsNA, 2018-02-19:
This issue will be fixed mechanically, at least for now it will be. I will keep this issue open until James gives the O.K. that the issue was fixed mechanically.

lindsayvallen, 2018-03-21:
Not yet fixed mechanically so fixed in software for auto moves.  Sets wrist to mid-up position before dropping elevator and then setting wrist to up position in ResetElevatorandWrist.  Before that's called, added back up to auto moves that needed it so they don't run the wrist/maw into anything when the elevator comes down.  See commit 2853022.

closed 2018-03-21 by lindsayvallen