first95/FRC2018 #64: Intermittent Elevator-Doesn't-Go-Down

jch13, 2018-02-20:
Elevator does not re-home to the bottom position reliably.

jwalthour, 2018-02-20:
It sounds like the Talon's perception of the 0  position has probably crept upward along the elevator.

I think this may be something we can fix by:
1. Remove the soft limit on the bottom of travel, so that the elevator can go down to the homing switch
2. Add logic such that, while the homing switch is tripped, we don't command the elevator to go any farther down.  (basically writing our own logic for a limit switch)

There might also be another path that involves wiring the homing switch into the Talon, and use the Talon's internal support for limit switches.  We would need to do some investigation, however, to see if that stops us from using the switch as a homing switch at the same time.

jch13, 2018-02-20:
Related- Elevator down seems to run slower than up, in general, which is weird. We will investigate hardware causes today.

jwalthour, 2018-02-20:
a364342b9a174ec2ee9c1707a8a64bb475231d38

closed 2018-02-20 by jwalthour